microsoft/pxt-adafruit #880: Can't round-trip the Colors enum

guillaumejenkins, 2018-11-20:
[image]

samelhusseini, 2018-11-20:
There's also an issue with decompiling Color enum. eg: 
```
    light.setAll(Colors.Black)
```

samelhusseini, 2018-11-20:
Not a regression for light.setAll(Colors.Black)

guillaumejenkins, 2018-11-20:
Yes, that's what I meant, the Colors enum can't be decompiled

pelikhan, 2019-03-12:
Crashes the decompiler in latest pxt.
```
TypeError: Cannot read property 'getTopBlocks' of null
    at http://localhost:3232/blb/pxtblockly.js:5518:35
    at Array.forEach (<anonymous>)
    at alignBlocks (http://localhost:3232/blb/pxtblockly.js:5512:22)
    at Object.patchBlocksFromOldWorkspace (http://localhost:3232/blb/pxtblockly.js:5494:17)
    at http://localhost:3232/blb/main.js:7053:44
From previous event:
    at Object.decompileAsync (http://localhost:3232/blb/main.js:7050:10)
    at http://localhost:3232/blb/main.js:13087:41
From previous event:
    at http://localhost:3232/blb/main.js:13085:22
From previous event:
    at http://localhost:3232/blb/main.js:13077:16
From previous event:
    at Editor.openBlocks (http://localhost:3232/blb/main.js:13016:35)
    at ProjectView.openBlocks (http://localhost:3232/blb/main.js:504:29)
    at MainMenu.openBlocks (http://localhost:3232/blb/main.js:7805:27)
    at HTMLUnknownElement.callCallback (http://localhost:3232/blb/main.js:64212:14)
    at Object.invokeGuardedCallbackDev (http://localhost:3232/blb/main.js:64250:16)
    at Object.invokeGuardedCallback (http://localhost:3232/blb/main.js:64299:29)
    at Object.invokeGuardedCallbackAndCatchFirstError (http://localhost:3232/blb/main.js:64313:43)
    at executeDispatch (http://localhost:3232/blb/main.js:64578:19)
    at executeDispatchesInOrder (http://localhost:3232/blb/main.js:64600:5)
    at executeDispatchesAndRelease (http://localhost:3232/blb/main.js:64698:5)
    at executeDispatchesAndReleaseTopLevel (http://localhost:3232/blb/main.js:64709:10)
    at forEachAccumulated (http://localhost:3232/blb/main.js:64679:8)
    at runEventsInBatch (http://localhost:3232/blb/main.js:64840:5)
    at runExtractedEventsInBatch (http://localhost:3232/blb/main.js:64849:3)
    at handleTopLevel (http://localhost:3232/blb/main.js:68306:5)
    at batchedUpdates (http://localhost:3232/blb/main.js:76618:14)
    at batchedUpdates (http://localhost:3232/blb/main.js:66051:12)
    at dispatchEvent (http://localhost:3232/blb/main.js:68387:5)
    at interactiveUpdates (http://localhost:3232/blb/main.js:76673:14)
pxt.reportException @ pxtapp.js:1858
pxtapp.js:1858 TypeError: Cannot read property 'getTopBlocks' of null
    at http://localhost:3232/blb/pxtblockly.js:5518:35
    at Array.forEach (<anonymous>)
    at alignBlocks (http://localhost:3232/blb/pxtblockly.js:5512:22)
    at Object.patchBlocksFromOldWorkspace (http://localhost:3232/blb/pxtblockly.js:5494:17)
    at http://localhost:3232/blb/main.js:7053:44
From previous event:
    at Object.decompileAsync (http://localhost:3232/blb/main.js:7050:10)
    at http://localhost:3232/blb/main.js:13087:41
From previous event:
    at http://localhost:3232/blb/main.js:13085:22
From previous event:
    at http://localhost:3232/blb/main.js:13077:16
From previous event:
    at Editor.openBlocks (http://localhost:3232/blb/main.js:13016:35)
    at ProjectView.openBlocks (http://localhost:3232/blb/main.js:504:29)
    at MainMenu.openBlocks (http://localhost:3232/blb/main.js:7805:27)
    at HTMLUnknownElement.callCallback (http://localhost:3232/blb/main.js:64212:14)
    at Object.invokeGuardedCallbackDev (http://localhost:3232/blb/main.js:64250:16)
    at Object.invokeGuardedCallback (http://localhost:3232/blb/main.js:64299:29)
    at Object.invokeGuardedCallbackAndCatchFirstError (http://localhost:3232/blb/main.js:64313:43)
    at executeDispatch (http://localhost:3232/blb/main.js:64578:19)
    at executeDispatchesInOrder (http://localhost:3232/blb/main.js:64600:5)
    at executeDispatchesAndRelease (http://localhost:3232/blb/main.js:64698:5)
    at executeDispatchesAndReleaseTopLevel (http://localhost:3232/blb/main.js:64709:10)
    at forEachAccumulated (http://localhost:3232/blb/main.js:64679:8)
    at runEventsInBatch (http://localhost:3232/blb/main.js:64840:5)
    at runExtractedEventsInBatch (http://localhost:3232/blb/main.js:64849:3)
    at handleTopLevel (http://localhost:3232/blb/main.js:68306:5)
    at batchedUpdates (http://localhost:3232/blb/main.js:76618:14)
    at batchedUpdates (http://localhost:3232/blb/main.js:66051:12)
    at dispatchEvent (http://localhost:3232/blb/main.js:68387:5)
    at interactiveUpdates (http://localhost:3232/blb/main.js:76673:14)
pxt.reportException @ pxtapp.js:1858
[... 42 lines cut ...]
Promise._settlePromiseFromHandler @ bluebird.js:3278
Promise._settlePromise @ bluebird.js:3335
Promise._settlePromise0 @ bluebird.js:3380
Promise._settlePromises @ bluebird.js:3459
(anonymous) @ bluebird.js:175
Promise.then (async)
schedule @ bluebird.js:4484
Async.settlePromises @ bluebird.js:174
Promise._fulfill @ bluebird.js:3406
Promise._resolveCallback @ bluebird.js:3198
Promise._settlePromiseFromHandler @ bluebird.js:3290
Promise._settlePromise @ bluebird.js:3335
Promise._settlePromise0 @ bluebird.js:3380
Promise._settlePromises @ bluebird.js:3459
(anonymous) @ bluebird.js:175
Promise.then (async)
schedule @ bluebird.js:4484
Async.settlePromises @ bluebird.js:174
Promise._fulfill @ bluebird.js:3406
Promise._resolveCallback @ bluebird.js:3198
(anonymous) @ bluebird.js:3250
pendingMsgs.(anonymous function) @ pxtapp.js:12838
recvHandler @ pxtapp.js:12822
worker.onmessage @ pxtapp.js:12851
```

closed 2019-03-12 by pelikhan