microsoft/pxt-adafruit #881: Max volume has been reduced in /beta

samelhusseini, 2018-11-20:
**Describe the bug**
This bug only affects beta but the max volume has been reduced (the whole thing is much quieter). 

**To Reproduce**
Steps to reproduce the behavior:
1. Try setVolume(256) on both beta and release and notice the difference on hardware

**Expected behavior**
Same behavior as release

samelhusseini, 2018-11-20:
@mmoskal any idea what could be going on here? Was this code touched at all?

pelikhan, 2018-11-20:
We changed the amp pin here: https://github.com/Microsoft/pxt-common-packages/commit/2e3ab195375de66909215d28d89e3429169a5d1f#diff-cbd245b705dd002eb55b88222b7aaf2f

samelhusseini, 2018-11-20:
Do we need to update that somewhere?

samelhusseini, 2018-11-20:
https://github.com/Microsoft/pxt-adafruit/blob/master/libs/circuit-playground/config.ts#L13
?

samelhusseini, 2018-11-20:
Any other pin mapping that's changed?

samelhusseini, 2018-12-05:
This is fixed.

closed 2018-12-05 by samelhusseini