microsoft/pxt-adafruit #886: showRing and setAll are incompatible

samelhusseini, 2018-11-21:
**Describe the bug**
It appears showRing and setAll are incompatible.

Try the following code: 
```
input.buttonA.onEvent(ButtonEvent.Click, function () {
    light.setAll(0x0000ff)
    pause(1000)
})
forever(function () {
    light.showRing(
    "red red red red red red red red red red"
    )
})
```

It doesn't work as expected which is show blue on button pressed, and red all the other times.
I think it has something to do with autoShow and how it determines what the buffer is.

@pelikhan

pelikhan, 2018-11-21:
this is expected because setAll is not an animation. It does not get queued. As soon as it hits pause, the forever handler runs.

samelhusseini, 2018-11-21:
Not sure that's expected from a user perspective.. Thoughts on how we can make it better?