microsoft/pxt-adafruit #890: Serial Read String function freezes CPX

bborncr, 2018-11-23:
Receiving data using the Serial Read String function freezes the CPX. Microsoft MakeCode for Circuit Playground Express.

**To Reproduce**
Steps to reproduce the behavior:
1. Add the Serial extension
2. Upload Test code
``` javascript
input.buttonA.onEvent(ButtonEvent.Click, function () {
    serial.writeLine("Hello")
})
forever(function () {
    console.log(serial.readString())
})
``` 
3. Press the A button on the CPX to verify that the Serial Write function is working
4. Send some data to the CPX over TX/RX (A7/A6). Have tried sending data using different line endings.
 
**Expected behavior**
The sent data should appear in the console. Pressing the A Button again should send "Hello" however the CPX hangs until reset.

**Desktop**
 - OS: Windows 10
 - Application: Microsoft MakeCode for Circuit Playground Express

abchatra, 2018-12-17:
@jamesadevine Try out this with your changes.

jamesadevine, 2018-12-18:
@abchatra Is @bborncr using Maker? Or are you asking me to test if I fixed it in Maker?

bborncr, 2018-12-18:
I tested Serial on maker.makecode.com with the CPX and send/receive works as expected:
``` javascript
let count = 0
forever(function () {
    serial.writeNumber(count)
    serial.writeLine("")
    count += 1
    if (serial.readString() == "abc") {
        serial.writeLine("Received!")
    }
    pause(1000)
})
```
I assume that makecode.adafruit.com and the Windows app haven't been updated yet because they both still lock on Serial receive.
@jamesadevine Thanks!

jamesadevine, 2018-12-18:
Thanks for testing @bborncr :smile: glad it works!

closed 2019-03-12 by pelikhan