microsoft/pxt-adafruit #905: Accessing IR Led from TypeScript not performant enough

BartAdv, 2018-11-27:
While trying to implement a custom IR protocol without resorting to C++, I've found out the results are unreliable compared to counterpart implemented in C++.

**To Reproduce**

Following MakeCode program:

```
const CYCLE = 1000000 / 38000
const HALF_PERIOD = CYCLE / 2

let irLed = pins.IROut

function sendMark(): void {
    for (let i = 0; i < 6; i++) {
        irLed.digitalWrite(true)
        control.waitMicros(HALF_PERIOD)
        irLed.digitalWrite(false)
        control.waitMicros(HALF_PERIOD)
    }
}

input.buttonA.onEvent(ButtonEvent.Click, function () {
    sendMark()
    control.waitMicros(500)
    sendMark()
    control.waitMicros(200)
    sendMark()
    control.waitMicros(100)
})
```

when tested with IR receiver (TSOP4838) hooked to Arduino running the Raw IR decoder sketch: https://learn.adafruit.com/ir-sensor/using-an-ir-sensor gives unstable results:

```
Received:
OFF     ON
2808 usec, 280 usec
int IRsignal[] = {
// ON, OFF
pulseIR(280);

Received:
OFF     ON
19940 usec, 420 usec
580 usec, 360 usec
320 usec, 280 usec
int IRsignal[] = {
// ON, OFF
pulseIR(420);
delayMicroseconds(580);
pulseIR(360);
delayMicroseconds(320);
pulseIR(280);

Received:
OFF     ON
37336 usec, 120 usec
int IRsignal[] = {
// ON, OFF
pulseIR(120);
```

The above shows readings for button pressed three times. It seems like decoder is unable to find out the HIGH occurences, which suggest that alternating the pin value with waiting the half-cycle amount of microseconds is not stable enough to be picked by the IR receiver?

kuch68, 2019-01-08:
Does the above makecode program result in an error free execution? In the Adafruit CPx editor I get an error when button A is pressed during the simulation: "Cannot read property DigitalWritePin of null" 

Has there been a change in the pxt that no longer exposes the IR pins, or have I not initialized something correctly?

pelikhan, 2019-01-08:
The simulator probably does not support those.

pelikhan, 2019-03-12:
Simulator fixed in nombed branch.

closed 2019-03-12 by pelikhan