microsoft/pxt-adafruit #907: Sound level rotate neopixel adds unexpected colors

jedgarpark, 2018-11-28:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jedgarpark, 2018-11-29:
no, I added a 2 second delay and it still is having a wild color party.

pelikhan, 2019-03-12:
Fixed by https://github.com/Microsoft/pxt-common-packages/pull/743/files

closed 2019-03-12 by pelikhan