microsoft/pxt-adafruit #924: Simple Crickit servo example freezes with neopixel 'error code'?

ladyada, 2018-12-06:
this one is something that got introduced recently perhaps. it definately didn't used to happen - moving a servo back and forth on crickit will crash and the neopixels on CPX will display a blue/red pattern (despite not having any neopixel code in the sketch)

see this post
https://forums.adafruit.com/viewtopic.php?f=64&t=144212

the code used to run before, now it doesnt run more than a few seconds
https://makecode.adafruit.com/94109-45880-18074-48992

jamesadevine, 2018-12-06:
@ladyada I'm working on hopefully making a more maintainable, less buggy build for CPX without mbed.

I did however take a cursory glance, and in the simulator I get a PANIC! 108 error message. If it's simulating this, there's a good change this will be happening on the real device.

Perhaps @pelikhan and @mmoskal could lend their thoughts?

ladyada, 2018-12-06:
yes thank you for the CPX rewrite! it will be awesome

the PANIC is a known 'bug', due to the simulator asking over i2c for a version number and not getting a reply (since its, well, simulated) - that's not related. its odd because jp does not get this when he compiles the example, and i do... and we're running the same version. uf2's in the thread link above

mmoskal, 2018-12-06:
The neopixel stuff is a panic message - it says `0x0000010100 == 20`. Panic 20 is out of memory.

ladyada, 2018-12-06:
ok maybe its because im using firefox to compile? will try on chrome

ladyada, 2018-12-06:
nope still doesnt work when i use chrome to compile. maybe cause im on window and he's on a mac? its really mysterious

ladyada, 2018-12-07:
its definitely a browser-based thing
https://forums.adafruit.com/viewtopic.php?f=64&t=144212&p=714014#p714014

jedgarpark, 2019-01-08:
This issue is coming up again https://forums.adafruit.com/viewtopic.php?f=64&t=145923 Happening for me on a Mac with Google Chrome 71.0.3578.98 (Official Build) (64-bit)

caternuson, 2019-01-10:
Another one:
https://forums.adafruit.com/viewtopic.php?f=47&t=145851

I tested a recreated on Ubuntu 18.04.1 LTS with Firefox 64.0 (64-bit).

caternuson, 2019-01-28:
https://forums.adafruit.com/viewtopic.php?f=58&t=146779

ladyada, 2019-01-29:
@pelikhan @mmoskal @jamesadevine @Jaqster  hiya folks - this bug is making crickit tough to use in makecode cause servos is the first thing people want to use and it hard crashes after a few minutes. can we help in any way to get this bug squished?

pelikhan, 2019-01-29:
Did a quick test on maker, and it seems that we are crashing earlier in there. 
https://makecode.com/_EDhcFa6REJh3

pelikhan, 2019-02-04:
It seems that this is due to using SPI to drive the neopixels.

ladyada, 2019-02-04:
are you using SPI DMA?

pelikhan, 2019-02-04:
Yes

ladyada, 2019-02-04:
kk what about this demo from original post? doesnt have any neopixels in it
https://makecode.adafruit.com/94109-45880-18074-48992

pelikhan, 2019-02-05:
@ladyada i am snowed in without a power supply for the crickit. Could you try this one? I disabled neopixels over SPI in the project settings. The light pattern is gone (and neopixels are still working).

https://makecode.com/_Uf0L7w27J496

pelikhan, 2019-02-05:
Found a cable... servo is moving fine and no more neopixel noise with the fix. I will push it to beta.

pelikhan, 2019-02-05:
Fix live in /beta. Make sure **v1.2.12** or higher is loaded (and this build is done https://travis-ci.org/Microsoft/pxt-adafruit/builds/489246261). @ladyada

TheKitty, 2019-02-05:
Just tested with a real CPX and Crickit. No dice with beta 2.1.12. It runs the code for like 2 loops and on the third one it stops and displays neopixel pattern blue-blue-red-blue-red-5 blues.

Image of About box:
[image]

TheKitty, 2019-02-05:
I tried the javascript program @pelikhan an wrote on 1.2.11, it works
[image]

on 1.2.12, the javascript will not compile, error 
[image]

And the .uf2 does nothing

TheKitty, 2019-02-05:
> @ladyada i am snowed in without a power supply for the crickit. Could you try this one? I disabled neopixels over SPI in the project settings. The light pattern is gone (and neopixels are still working).
> 
> https://makecode.com/_Uf0L7w27J496

This code does not compile, 3 errors. Taking an extra ) off the end takes it to one error but I cannt determine what that is

pelikhan, 2019-02-05:
Just tried this sample and it worked for me on HW.
```
let index = 0
crickit.servo1.setAngle(0)
forever(function () {
    for (let index = 0; index <= 18; index++) {
        crickit.servo1.setAngle(index * 10)
        pause(100)
    }
    for (let index = 0; index <= 18; index++) {
        crickit.servo1.setAngle(180 - index * 10)
        pause(100)
    }
    pause(1000)
})
forever(function () {
    light.showAnimation(light.rainbowAnimation, 500)
})
```

pelikhan, 2019-02-05:
@TheKitty you need to add the "crickit" extension.

TheKitty, 2019-02-05:
That makes sense, just a sec

TheKitty, 2019-02-05:
Adding the Crickit extension, your code works on 1.2.12 beta

TheKitty, 2019-02-05:
I am retrying a block based program on the same make. The CPX will not go into bootloader mode. I will look to put circuitpython on it then put into bootloader mode then try a block program

TheKitty, 2019-02-05:
Yes, right now, my CPX is essentially stuck running the javascript code. I cannot get into bootloader no matter what I try. Conferred with Limor.

jedgarpark, 2019-02-08:
The issue (or some version of it?) seems to still be occurring in the example from [this forum post](https://forums.adafruit.com/viewtopic.php?f=64&t=145923&p=727995#p721291):
```
let notes: number[] = []
let i = 0
let open_angle = 0
let close_angle = 0
let lvl = 0
let angle = 0
/**
 * An example of playing two loops concurrently. This based on Snake Charmer Box and Jingle Bells With Red-Green Lights
 */
/**
 * Crashes shortly after beginning 2nd pin A1 touch!
 */
input.pinA1.onEvent(ButtonEvent.Click, function () {
    crickit.servo1.setAngle(close_angle)
    pause(100)
    move_servo_and_play_notes2()
})
function move_servo_and_play_notes2() {
    angle = open_angle
    i = 0
    while (angle > close_angle && angle <= open_angle) {
        if (i > notes.length - 1) {
            i = 0
        }
        if (notes[i] != 0) {
            music.playTone(notes[i], music.beat(BeatFraction.Half))
        } else {
            music.rest(music.beat(BeatFraction.Half))
        }
        angle = angle - 2
        crickit.servo1.setAngle(angle)
        pause(100)
        i = i + 1
    }
}
lvl = 32
close_angle = 30
open_angle = 150
notes = [262, 294, 311, 294, 262, 0, 262, 294, 311, 392, 294, 311, 262, 0, 0]
```
I wonder if this is due to the addition of sound to the mix?

jedgarpark, 2019-02-08:
@pelikhan do you want to reopen this issue?

pelikhan, 2019-02-08:
Oh boy... here we go again.

jedgarpark, 2019-02-08:
Wheeeeee

tavdog, 2019-02-14:
I'm seeing this issue when running a motor.  The code works for a few loops then crashes. Using CPX and Crickit with Adafruit geared motors.

pambrose, 2019-02-15:
@tavdog If you are on OSX, make sure you are using Chrome and not Safari.

tavdog, 2019-02-15:
Yes I am using Chrome on OSX 10.11.16

danowall, 2019-02-21:
I'm seeing a crashing issue when I run the code below

https://makecode.adafruit.com/36105-40587-05667-52989

This is set up so that button presses trigger servos to move, it works for at most about a minute before crashing. There is no change to any onboard NeoPixels or LEDs when it crashes. I've noted the behavior is the same whether using the regular /#editor or /beta#editor suffix. It will always eventually crash independent of whether or not any buttons are pressed. 

Using Chrome on a Macbook Air running OSX 10.14.3
Code written with MakeCode version 5.1.5 for Adafruit Circuit Playground Express version 1.2.12

[image]

[image]

pelikhan, 2019-03-07:
Investigating https://github.com/Microsoft/pxt-adafruit/issues/924#issuecomment-461688664

pelikhan, 2019-03-07:
No lock observed on maker (using new CODAL) - https://makecode.com/_Dru2k6iayDgp

ladyada, 2019-03-09:
one thing to really try a bunch is reading from the crickit, capacitive touch and the 8-pin 'I/O' port never worked :(

pelikhan, 2019-03-09:
You mean capacitive touch from crickit?

ladyada, 2019-03-09:
yep

pelikhan, 2019-03-12:
@ladyada we don't have the bandwidth to test the crickit features. Looks like it is fixed on this end at least. Tested on the nombed branch.

closed 2019-03-12 by pelikhan

ladyada, 2019-03-12:
ok we can do the testing and report back! fingers crossed :)

tavdog, 2019-03-12:
I will test to see if this fixes the crashes I've been seeing ? Is the fix live on makecode.com ?

pelikhan, 2019-03-12:
@tavdog @ladyada @danowall @jedgarpark @TheKitty @jamesadevine all the fixes are available in https://makecode.adafruit.com/beta (v1.3.*)