microsoft/pxt-adafruit #925: Upload Cartoon Network Projects

Jaqster, 2018-12-17:
Hey Galen,
Can you get these Cartoon Network projects propped?  We're planning to go live with them on January 7th on the Home Page, but let's start putting them up now.
Here's the first 4 - https://1drv.ms/f/s!AvnYExTCocXPhldY1V6FzliyZB-T
I'll keep adding to these and let you know when a new one is ready.
Let me know if you have any questions.
Thanks!
Jacqueline

ganicke, 2018-12-17:
Ok, I'll put them in. We'll just "turn on" the Cartoon Network gallery in the Home Page on 1/7.

Jaqster, 2018-12-28:
Also noticed a small typo in the materials bulleted List for the Finn Cup Lamp project - https://makecode.adafruit.com/projects/cartoon-network/cup-lamp

Jaqster, 2019-01-03:
Hey Galen,
Just heard from CN - they want to use the project photos instead of the cartoon images for the Cards.  Here's the images they want to use.
[Project Images.zip](https://github.com/Microsoft/pxt-adafruit/files/2725283/Project.Images.zip)

Jaqster, 2019-01-03:
Also, they sent me some higher resolution images to use in the project write-ups.  I've posted them here: https://1drv.ms/f/s!AvnYExTCocXPiHMyfvxMKHzPBIhO

ganicke, 2019-01-03:
Ok, thanks. I'll get them in there.

ganicke, 2019-01-17:
All 7 CN projects are published.

closed 2019-01-17 by ganicke