microsoft/pxt-adafruit #937: Cartoon Network - Go Live stuff

Jaqster, 2019-01-03:
Here's the stuff that needs to go live for the Cartoon Network launch on Monday (1/7).
[ ] New Home Page banner (files at https://github.com/Microsoft/pxt-adafruit/tree/master/resources)
[ ] New "Cartoon Network" carousel on Home Page with project cards (let's put this on the first row for a few weeks, and then we can move it down)
[ ] Blog post - I'll work on a draft of this

Jaqster, 2019-01-04:
Here are the files for the Blog post.  Feel free to edit.

[Cartoon Network Announce.docx](https://github.com/Microsoft/pxt-adafruit/files/2728472/Cartoon.Network.Announce.docx)
[image]
[image]

ganicke, 2019-01-04:
@abchatra added gallery and banner in #939

ganicke, 2019-01-17:
CN stuff posted and live.

closed 2019-01-17 by ganicke