microsoft/pxt-adafruit #943: IR causing Music to misbehave

jedgarpark, 2019-01-24:
**Describe the bug**
Adding IR send and receive to code causes music to play back extremely slowly.

**To Reproduce**
`/**
 * Garnet's Palm Gems Duo Version!
 * 
 * Fusion of bringing together Ruby (red) and Sapphire (blue) to create Garnet (pink)
 * 
 * Press B to do the fusion.
 * 
 * Press A to un-fuse.
 */
network.onInfraredReceivedNumber(function (num) {
    if (num == 0) {
        light.showAnimation(light.sparkleAnimation, 300)
        bleepSong2()
        light.showRing(
        `red red red red red blue blue blue blue blue`
        )
    } else {
        light.showAnimation(light.cometAnimation, 1500)
        light.showRing(
        `pink pink pink pink pink pink pink pink pink pink`
        )
        themeSong2()
        light.setBrightness(120)
        pause(500)
        light.setBrightness(30)
    }
})
input.buttonA.onEvent(ButtonEvent.Click, function () {
    network.infraredSendNumber(0)
    light.showAnimation(light.sparkleAnimation, 300)
    bleepSong2()
    light.showRing(
    `red red red red red blue blue blue blue blue`
    )
})
input.buttonB.onEvent(ButtonEvent.Click, function () {
    network.infraredSendNumber(1)
    light.showAnimation(light.cometAnimation, 1500)
    light.showRing(
    `pink pink pink pink pink pink pink pink pink pink`
    )
    themeSong2()
    light.setBrightness(120)
    pause(500)
    light.setBrightness(30)
})
function themeSong2() {
    music.playTone(262, music.beat(BeatFraction.Whole))
    music.playTone(440, music.beat(BeatFraction.Quarter))
    music.playTone(392, music.beat(BeatFraction.Eighth))
    music.playTone(349, music.beat(BeatFraction.Quarter))
    music.playTone(392, music.beat(BeatFraction.Quarter))
    music.rest(music.beat(BeatFraction.Eighth))
    music.playTone(330, music.beat(BeatFraction.Half))
    music.playTone(330, music.beat(BeatFraction.Quarter))
    music.playTone(277, music.beat(BeatFraction.Quarter))
    music.playTone(440, music.beat(BeatFraction.Quarter))
    music.playTone(392, music.beat(BeatFraction.Quarter))
    music.playTone(349, music.beat(BeatFraction.Quarter))
    music.playTone(330, music.beat(BeatFraction.Eighth))
    music.playTone(349, music.beat(BeatFraction.Eighth))
    music.playTone(349, music.beat(BeatFraction.Quarter))
    music.playTone(349, music.beat(BeatFraction.Eighth))
    music.rest(music.beat(BeatFraction.Quarter))
    music.rest(music.beat(BeatFraction.Eighth))
    music.playTone(262, music.beat(BeatFraction.Quarter))
    music.playTone(440, music.beat(BeatFraction.Quarter))
    music.playTone(392, music.beat(BeatFraction.Quarter))
    music.playTone(349, music.beat(BeatFraction.Eighth))
    music.playTone(330, music.beat(BeatFraction.Quarter))
    music.playTone(349, music.beat(BeatFraction.Eighth))
    music.playTone(349, music.beat(BeatFraction.Eighth))
    music.rest(music.beat(BeatFraction.Quarter))
    music.rest(music.beat(BeatFraction.Eighth))
    music.playTone(262, music.beat(BeatFraction.Quarter))
    music.rest(music.beat(BeatFraction.Quarter))
    music.playTone(440, music.beat(BeatFraction.Quarter))
    music.playTone(392, music.beat(BeatFraction.Quarter))
    music.playTone(349, music.beat(BeatFraction.Quarter))
    music.playTone(330, music.beat(BeatFraction.Eighth))
    music.playTone(349, music.beat(BeatFraction.Eighth))
    music.playTone(349, music.beat(BeatFraction.Quarter))
    music.playTone(349, music.beat(BeatFraction.Eighth))
}
function bleepSong2() {
    music.playTone(523, music.beat(BeatFraction.Sixteenth))
    music.playTone(784, music.beat(BeatFraction.Sixteenth))
    music.playTone(880, music.beat(BeatFraction.Sixteenth))
    music.playTone(523, music.beat(BeatFraction.Sixteenth))
    music.playTone(784, music.beat(BeatFraction.Sixteenth))
    music.playTone(880, music.beat(BeatFraction.Sixteenth))
    music.playTone(523, music.beat(BeatFraction.Sixteenth))
    music.playTone(784, music.beat(BeatFraction.Sixteenth))
    music.playTone(880, music.beat(BeatFraction.Sixteenth))
    music.playTone(523, music.beat(BeatFraction.Sixteenth))
    music.playTone(784, music.beat(BeatFraction.Sixteenth))
    music.playTone(880, music.beat(BeatFraction.Sixteenth))
}
music.setTempo(52)
light.setBrightness(30)
light.showRing(
`blue blue blue blue blue red red red red red`
)
`

pelikhan, 2019-03-12:
@jamesadevine

pelikhan, 2019-03-12:
It does not crash but IR is very unreliable.

jamesadevine, 2019-03-12:
@pelikhan should be fixed now, was a small buggette in the TCC timer code.

closed 2019-03-12 by pelikhan