microsoft/pxt-adafruit #944: Add internal pull-up resistor support for Crickit Signal IO

jedgarpark, 2019-01-30:
The Signal IO ports currently can be read for digital IO, but require use of external pull-up resistors. Could we add support for enabling the internal pull-ups?

pelikhan, 2019-01-30:
Do you have more details? There is the pins.....setPull(...) api

pelikhan, 2019-03-07:
@jedgarpark does https://makecode.adafruit.com/reference/pins/set-pull work for you?

jedgarpark, 2019-03-08:
Thanks, that works perfectly. Now, to hunt down the original request on the Adafruit forums : ]

jedgarpark, 2019-03-08:
Oops, I misread that and only tested on CPX, my bad. Reopening while I test on Crickit.

jedgarpark, 2019-03-08:
I don't know how I would use **set pull pin** for a Crickit Signal pin.

pelikhan, 2019-03-08:
Oh, so that's really a request for the https://github.com/adafruit/pxt-crickit repo...

jedgarpark, 2019-03-08:
Ah, yes. Looks like it exists there: https://github.com/adafruit/pxt-crickit/issues/11

closed 2019-03-08 by jedgarpark