microsoft/pxt-adafruit #950: "crickit run motor" causes "PANIC! 108"

darzu, 2019-03-06:
**Describe the bug**
Using "cricket run motor" block causes the editor to error out with Code 108.

**To Reproduce**
1. Open: https://makecode.com/_3rF5Yz6oh6Ka
2. Notice the panic

**Expected behavior**
Don't panic.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 10.14.3 (18D109)
 - Browser Safari Version 12.0.3 (14606.4.5)
 - MacBook Pro 2018, i9, 32GB

pelikhan, 2019-03-06:
Adafruit designed the library so that it fails hard if the i2c handshake fails. We should probably make it fail silently.

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F6453828%2F53921201-dcb66800-406f-11e9-8bc9-a373f7e6668e.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C7754cf14b2f44aa2c71908d6a28b72d4%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636875117148120399&sdata=gRO8TSqRRU1UbWB6W0oI3k1CzYt7GoJsbe0lOZUmnB0%3D&reserved=0>

darzu, 2019-03-06:
Does that mean no cricket programs work in the simulator?

pelikhan, 2019-03-07:
Correct no simulator support for now. We don’t have a shield story really. Not crashing would be much better.

darzu, 2019-03-07:
Some sort of "sorry, this code can't run in the simulator" error might be nice.
But shouldn't we be able to run the rest of the code (e.g. if there are some neopixel animations) regardless?

pelikhan, 2019-03-07:
IMO we should silently ignore the failure. Or maybe blink the D13 LED.

pelikhan, 2019-03-12:
This issue belongs to the https://github.com/Adafruit/pxt-crickit repo.

closed 2019-03-12 by pelikhan