microsoft/pxt-adafruit #954: light sensor only reads value between 51 (e.g. 200) and 193 (eg ~800)

pelikhan, 2019-03-13:
It seems that the sensivity of the light sensor has changed. It does not go near zero when covered or does not max when pointing a phone flash light.

jamesadevine, 2019-03-13:
Hmmmm pretty sure I was seeing a far bigger range. Will take a look

jamesadevine, 2019-03-19:
https://github.com/lancaster-university/codal-circuit-playground/releases/tag/v1.5.4-samd-peripherals.11

pelikhan, 2019-03-19:
Nice work!

https://github.com/lancaster-university/codal-circuit-playground/releases/tag/v1.5.4-samd-peripherals.11<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Flancaster-university%2Fcodal-circuit-playground%2Freleases%2Ftag%2Fv1.5.4-samd-peripherals.11&data=02%7C01%7Cjhalleux%40microsoft.com%7Cc4886b710b704f25652008d6ac2973c9%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636885691368060220&sdata=txNZiTXnoZRQQ1pQydvJhc1q6nYknx3SoqVPO8Z6I%2BI%3D&reserved=0>

closed 2019-03-19 by pelikhan