microsoft/pxt-adafruit #955: Serial RX Events not working in non-mbed (/beta)

bborncr, 2019-03-13:
In the no-mbed branch (testing on makecode.adafruit.com/beta) the Serial RX events never trigger.

**Steps to reproduce the behavior**
1. Go to makecode.adafruit.com/beta
2. Use this test code:
```javascript
serial.onEvent(SerialEvent.DataReceived, function () {
    light.setAll(0x00ff00)
})
let counter = 0
serial.setBaudRate(BaudRate.BaudRate9600)
forever(function () {
    serial.writeNumber(counter)
    serial.writeLine("")
    counter += 1
    pause(1000)
})
```
3. Connect a UART to USB adaptor to the CPX on TX/RX (9600 baud in my case).

**Expected behavior**
The code sends a counter to TX every second. This works.
When sending any data to the CPX all the leds should turn green.

**CPX Version**
UF2 Bootloader v2.0.0-adafruit.9 SFHWRO

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version 72.0.3626.121 (Official Build) (64-bit)

pelikhan, 2019-03-19:
This issue is that i mistakenly used the RX BUFFER Full event. Tracking.

pelikhan, 2019-03-20:
Fixed in beta. Make sure to read in the event to avoid filling up the rx buffer.

closed 2019-03-20 by pelikhan

bborncr, 2019-03-21:
`serial.onEvent` tested and working (using Serial Terminal in Arduino IDE with line ending set to "Newline"):
```
let message = ""
serial.onEvent(SerialEvent.DataReceived, function () {
    message = serial.readUntil(Delimiters.NewLine)
    if (message == "hello") {
        light.setAll(0x00ff00)
    } else {
        light.setAll(0x000000)
    }
})
```
However, `serial.onDelimiterReceived` does not fire upon receiving a message with a Newline in this example:
```
serial.onDelimiterReceived(Delimiters.NewLine, function () {
    light.setAll(0x00ff00)
})
```