microsoft/pxt-adafruit #956: Light levels problems in no-mbed (/beta)

bborncr, 2019-03-13:
Light levels in the CPX are not calibrated correctly using the no-mbed code (/beta)
- Using finger to cover the light sensor = 44 (should be close to zero)
- Ambient light in my office = 59
- LED from phone = 194

I expect the values of the light sensor to be close to zero when the light sensor is covered and near the maximum (255?) when a bright light is shined directly on it.

If the light sensor does not go down close to zero then functions like graph() do not work properly.

bborncr, 2019-03-13:
Duplicate issue already reported by @pelikhan

closed 2019-03-13 by bborncr

pelikhan, 2019-03-13:
Your report was more detailed!

Closed #956<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMicrosoft%2Fpxt-adafruit%2Fissues%2F956&data=02%7C01%7Cjhalleux%40microsoft.com%7C4429a291f4e544c9852b08d6a7d2ecb9%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636880921685147993&sdata=h9UuxMYBV46boQ4krAvo1DpmyjM5ZMpaCzBHNLmURgU%3D&reserved=0>.