microsoft/pxt-adafruit #960: "MUSIC" drawer doesn't show in toolbox of "Siren" tutorial

Helen09, 2019-03-18:
**Describe the bug**
"MUSIC" drawer doesn't show in toolbox of "Siren" tutorial

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/beta
2. Open tutorial "Siren" and click tutorial step one by one
3. When clicking to 5th step observe the toolbox

**Expected behavior**
"MUSIC" drawer should show in toolbox

**Actual behavior**
"MUSIC" drawer doesn't show in toolbox of "Siren" tutorial
[image]

**Additional info:**
1. This bug repro on Edge, Firefox, Chrome & IE on Windows
(17763.1.amd64fre.rs5_release.180914-1434)

pelikhan, 2019-03-19:
Requires to update the tutorial to reflect the new music blocks.

pelikhan, 2019-03-20:
Waiting for release to fix.

abchatra, 2019-04-26:
@ganicke we are planning to release this today. Sorry for the late notice. Is it possible to generate a PR for these tutorials. As soon as the release is done, we will merge the PR.

PR #1016 (2019-04-26): Update tutorials for melody play
Change tutorials with ``playSound()`` to ``play()`` for fixed melody instances.

Fixes #960 
Fixes #961
Fixes #985
closed 2019-04-26 by abchatra