microsoft/pxt-adafruit #961: "MUSIC" drawer doesn't show in toolbox of "Magic wand" tutorial

Helen09, 2019-03-18:
**Describe the bug**
"MUSIC" drawer doesn't show in toolbox of "Magic wand" tutorial

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/beta
2. Open tutorial "Magic Wand" and click tutorial step one by one
3. When clicking to 4th step observe the toolbox

**Expected behavior**
"MUSIC" drawer should show in toolbox

**Actual behavior**
"MUSIC" drawer doesn't show in toolbox of "Magic wand" tutorial

**Additional info:**
This bug repro on Edge, Firefox, Chrome & IE on Windows
(17763.1.amd64fre.rs5_release.180914-1434)

Helen09, 2019-03-18:
This bug also repro on tutorial "Button flipper"'s 5th step, there is no "MUSIC" drawer in toolbox
[image]

pelikhan, 2019-03-19:
Requires updates to tutorial to match new music blocks.

pelikhan, 2019-03-20:
Waiting for release to fix.

abchatra, 2019-04-26:
@ganicke we are planning to release this today. Sorry for the late notice. Is it possible to generate a PR for these tutorials. As soon as the release is done, we will merge the PR.

PR #1016 (2019-04-26): Update tutorials for melody play
Change tutorials with ``playSound()`` to ``play()`` for fixed melody instances.

Fixes #960 
Fixes #961
Fixes #985
closed 2019-04-26 by abchatra