microsoft/pxt-adafruit #962: There should only one 5th step in tutorial "room-sign"

Helen09, 2019-03-18:
**Describe the bug**
There should only one 5th step in tutorial "room-sign"

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/projects/cartoon-network/room-sign
2. Scroll down to the 5th step and observe

**Expected behavior**
There are duplicate 5th step in tutorial "room-sign"

**Actual behavior**
There should only one 5th step in tutorial "room-sign"
[image]

Helen09, 2019-03-18:
This bug also repro on https://makecode.adafruit.com/projects/cartoon-network/bmo-music-box
The step is from step 4 turns to step 7
[image]

pelikhan, 2019-03-18:
@ganicke

ganicke, 2019-03-18:
Well, that's an oops...will fix that.

closed 2019-03-18 by pelikhan

Helen09, 2019-03-19:
@ganicke This bug also repro on  https://makecode.adafruit.com/projects/cartoon-network/bmo-music-box

ganicke, 2019-03-19:
Thanks @Helen09 I see the step order problem. Will fix.

PR #964 (2019-03-18): Fix order of Room Sign make steps
Fix repeated step number in Room Sign make page.

Fixes #962