microsoft/pxt-adafruit #967: Issue on tutorial Rube Goldberg

Helen09, 2019-03-19:
**Describe the bug**
Issue on tutorial Rube Goldberg

**Steps to reproduce the behavior**
1. Open https://makecode.adafruit.com/projects/rube-goldberg/code
2. Scroll down this page and observe step 3 and step 4

**Expected result:**
Tutorial should change the A0 in both "servo write pin" blocks to A2 because A3 can't be chosen 
[image]

**Actual result:**
Issue on tutorial Rube Goldberg
[image]

[image]

Helen09, 2019-03-19:
The 4th step also need to be updated
[image]

pelikhan, 2019-03-19:
@ganicke can you take a peek?

closed 2019-03-21 by pelikhan