microsoft/pxt-adafruit #976: There are duplicate 3rd step in tutorial "musical-cloud"

Helen09, 2019-03-20:
**Describe the bug**
There are duplicate 3rd step in tutorial "musical-cloud"

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/projects/musical-cloud/code
2. Scroll down to the 3rd step and observe

**Expected behavior**
There should only one 3rd step in tutorial "musical-cloud"

**Actual behavior**
There are duplicate 3rd step in tutorial "musical-cloud"
[image]

pelikhan, 2019-03-20:
@ganicke

ganicke, 2019-03-20:
Aey, this seems epidemic. ``pxt checkdoc --validate-steps``. Just kidding.

Helen09, 2019-03-21:
Closed as fixed

closed 2019-03-21 by Helen09