microsoft/pxt-adafruit #985: A "twinkle" variable from "variables" can't be put into "play sound"

Helen09, 2019-04-03:
**Describe the bug**
A "twinkle" variable from "variables" can't be put into "play sound"

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/projects/musical-cloud/code
2. Follow the step to drag blocks on content pane
3. Until the 4th step, as shared project
https://makecode.com/_FWRUhkY3tgj4

**Expected behavior**
Drag a "twinkle" variable from "variables" into the "play sound" where is says "power up"
[image]

**Actual behavior**
A "twinkle" variable from "variables" can't be put into "play sound"
[image]

**Additional info:**
1. test OS: Windows
2. Adafruit Circuit Playground Express version:  1.4.7
    Microsoft MakeCode version:  5.5.37
    codal-circuit-playground runtime version:  v1.5.4-samd-peripherals.11

pelikhan, 2019-04-17:
I think this is more of a block definition issue.

ganicke, 2019-04-18:
The situation is that this project wants to compose a sound and uses the now deprecated **playSound** block. The new melody blocks don't support a custom sound string. 

The new **play** block has he exact same text as **playSound**. So, the coding steps won't work with the available blocks. We'll need to get rid of the "twinkle" sound or have the user to stay in JS only.

```typescript
let twinkle = ""
music.playSound(twinkle)
music.baDing.play()
```

[image]

pelikhan, 2019-04-18:
The customizable api was nice. This is an issue.

ganicke, 2019-04-18:
Maybe a melody editor, makes one byte notes packed into `Buffer`... sometime later, over the horizon.

pelikhan, 2019-04-18:
It perfectly acceptable to instantiate a new Melody and play it. We should have a block that takes a string.

ganicke, 2019-04-18:
Yea, maybe add a helper to create it that takes a string: `let song = createMelody("e:2 b:4")`?

pelikhan, 2019-04-18:
Will not work with fixed instances...

ganicke, 2019-04-19:
Change melody up a bit to do...

```
//% fixedInstances
export class StaticMelody extends Melody {
}

export class ComposedMelody extends Melody {
}
```

...except compile the melody on create rather than in `play()` in order to not retain the bulky melody string.

abchatra, 2019-04-26:
@ganicke we are planning to release this today. Sorry for the late notice. Is it possible to generate a PR for these tutorials. As soon as the release is done, we will merge the PR.

abchatra, 2019-04-26:
Hmm ignore my comment. I don't know what we can do here.

ganicke, 2019-04-26:
OK, I'm fixing the others now...

PR #1016 (2019-04-26): Update tutorials for melody play
Change tutorials with ``playSound()`` to ``play()`` for fixed melody instances.

Fixes #960 
Fixes #961
Fixes #985
closed 2019-04-26 by abchatra