microsoft/pxt-adafruit #988: Cannot deploy on local host

suraj022, 2019-04-04:
I'm trying to install pxt-adafruit on my local system but when I try to deploy it using `pxt deploy` inside `libs/circuit-playground` folder, I get the following error:

`Failed to initialize libusb.
INTERNAL ERROR: Error: requestDevice error: no devices found
    at /home/ubuntu/pxt-adafruit/node_modules/webusb/lib/usb.js:196:35
    at tryCatcher (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/promise.js:512:31)
    at Promise._settlePromise (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/promise.js:569:18)
    at Promise._settlePromise0 (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/promise.js:614:10)
    at Promise._settlePromises (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/promise.js:693:18)
    at Async._drainQueue (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/async.js:133:16)
    at Async._drainQueues (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/async.js:143:10)
    at Immediate.Async.drainQueues [as _onImmediate] (/home/ubuntu/pxt-adafruit/node_modules/bluebird/js/release/async.js:17:14)
    at processImmediate (internal/timers.js:443:21)`

abchatra, 2019-04-15:
We don't support local host issues.

closed 2019-04-15 by abchatra

suraj022, 2019-04-16:
Same error occurs when I try this on AWS hosted system.