microsoft/pxt-adafruit #991: Adafruit CPX does not show console device option using WebUSB in MakeCode

FrankSchmidtDDS, 2019-04-12:
**Describe the bug**
A clear and concise description of what the bug is.
Unlike Micro:Bit using the WebUSB option which does show USB data feed back to the computer screen by way of the WebUSB editor in MakeCode, Circuit Playground Express does not give that function.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to : https://makecode.adafruit.com/beta?webusb=1#editor
2. Click on https://makecode.com/_E8eH2L5zA8aJ
But now you must go to the JavaScript tab of the above link, copy that JS code and then paste it into the JS tab on the #1. above web site, ?bet"webusb=1#editor. Do not use the non-webusb web site.
3. Once code is installed, and if there are no errors, plug in the CPX with USB cord and click on the gear icon upper left of the computer screen, choose "pair device" , if all LEDs are green for loading code, then highlight the text in the window and click "connect" when it is activated. Once that is done, click DOWNLOAD. 
4.At this point, I do not get the flickering upload light on the CPX. The code does not load. The second oval, below the simulator does not appear.

**Expected behavior**
I expected the code to upload to the CPX. It does not. Following the same procedure for the Micro:Bit does upload code. For the CPX the code just downloads to DOWNLOADS file as per normal, non-WebUSB download procedures. Furthermore , the second oval window below simulator which should have text reading: Show console device, does not appear. It does so with Micro:Bit. Then data can be read on the computer monitor coming from sensors. 

**Screenshots**
[image]

Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - Microsoft Windows 10 on Surface Pro and also on ASUS
Also Chrome OS on Samsung Chromebooks
Chrome browsers only, updated on all machines. 
Bootloader updated to v 2.0.0 on 2 different CPX s

10 different data cables used, all work for uploading code to all other processors. 

**Additional context**
Add any other context about the problem here.

JKCOS, 2019-06-16:
I am having the same problem.  Sometimes the Device Simulator launches when I download the file for the Playground Express but not usually so I am not able to see the serial feed from the board.

FrankSchmidtDDS, 2019-06-17:
Try this beta editor : https://makecode.adafruit.com/?webusb=1#editor

After you get to that URL page then click IMPORT and put in this URL share link: https://makecode.com/_4dER7s9CcT9s

Then plug in your CPX and click the gear icon upper right and click "Pair device" and see if that consistently shows the console device output to your computer screen. Let me know if it does not work for you.

JKCOS, 2019-06-17:
Yes, that did work. Thank you!

abchatra, 2019-09-03:
This should be fixed. If not let us know.

closed 2019-09-03 by abchatra