microsoft/pxt-adafruit #1001: Tutorial dialog rendering issues on Edge

Jaqster, 2019-04-19:
Saw this a few times clicking through the tutorials - 
 - The dialog box doesn't get repainted correctly
 - Sometimes the screen will freeze and you have to reload the page

[image]

[image]

Jaqster, 2019-04-19:
Also on Edge, sometimes it will accidentally select text for you just by mousing over things

[image]

pelikhan, 2019-04-19:
Test selection issue happening in all editors.