microsoft/pxt-adafruit #1005: ``show ring`` field editor doesn't work on iOS

jwunderl, 2019-04-19:
**Describe the bug**

``show ring`` field editor doesn't work at all on ipad on live site or beta:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to editor on ipad
2. Click on drag out show ring from light
3. try to change light colors
4. See error

**Expected behavior**

lights in the ring should change colors

**Smartphone (please complete the following information):**
 - Device: iPad pro 12.9 in 3rd gen
 - OS: iOS

abchatra, 2019-04-19:
This is not a blocking bug.

pelikhan, 2019-04-26:
Broken on iPhone as well.

mlewisosu, 2019-10-10:
While this wasn't working for me either in iOS 12.x it seems to be fixed now in iOS 13.1.2 (on iPhone)