microsoft/pxt-adafruit #1006: Importing old script

Jaqster, 2019-04-19:
Old block is not upgraded.
Import attached script, toggle to blocks.

[image]

[circuitplayground-guitar.uf2.txt](https://github.com/Microsoft/pxt-adafruit/files/3098998/circuitplayground-guitar.uf2.txt)

abchatra, 2019-04-25:
@jwunderl can you please help fix this for tomorrow?

jwunderl, 2019-04-25:
Looks like the upgrade rules are getting applied properly, and somehow just leaving a broken block in the xml that shouldn't be there; the original version (it seems to be from around v0.16.22) has the same contents

[image]

The upgrade rule is upgrading the music block from ``music_play_note`` to ``mixer_play_note`` properly, but not removing the old ``music_play_note`` and leaving it as 

```xml
  <block type="music_play_note" disabled="true" x="232" y="0"></block>
```

Looking into where that can be fixed now

closed 2019-08-15 by abchatra