microsoft/pxt-adafruit #1037: `Light` category has two subsections

jwunderl, 2019-05-23:
**Describe the bug**

Light section has both `neopixel` and `more` as subsections, instead of just `neopixel`. `more` has only a single block

**To Reproduce**

Steps to reproduce the behavior:
1. Go to beta editor
2. open light category
3. see two new categories instead of just one

**Expected behavior**

Block in more category should probably just be in light? Seems weird to me to have two categories pop up

**Screenshots**

[image]

abchatra, 2019-05-23:
Good bug

abchatra, 2019-05-23:
@pelikhan Is this your bug? Can you fix it?

pelikhan, 2019-05-23:
Yes that bloc needs a group on it.

jwunderl, 2019-05-23:
Hm, that block also just doesn't compile to javascript?

closed 2019-05-23 by abchatra