microsoft/pxt-adafruit #1038: Monaco flyout issues in portrait mode

jwunderl, 2019-05-23:
**Describe the bug**

in portrait  / low width modes where simulator is at the bottom, the monaco flyout can break slightly. If you switch to blocks in this view, the simulator will be half hidden until you open a flyout again and drag out a blck

**To Reproduce**

Steps to reproduce the behavior:
1. Go to monaco editor in beta
2. open a flyout
3. see error
4. switch to blocks; error persists

**Expected behavior**

simulator area should just be fully hidden / unhidden

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: mac
 - Browser chrome

**Smartphone (please complete the following information):**
 - Device: android emulator
 - Browser chrome

abchatra, 2019-05-23:
Is this localhost or beta? Please try beta.

jwunderl, 2019-05-23:
Yes, repros on beta