microsoft/pxt-adafruit #1041: Play Tone still sounds buzzy

Jaqster, 2019-05-23:
Audio is much improved!  
Although I'm noticing that while the Play Sound melodies all sound very nice, and volume control seems correct, the Play Tone blocks still produce sound that is very "buzzy".  
Also, when I put a Play Sound block in a loop, a strange noise sounds between sounds.
https://makecode.com/_4igATVd3K7xe

abchatra, 2019-05-23:
@mmoskal

mmoskal, 2019-05-23:
@Jaqster does this sound less buzzy? (the weird sound between playUntilDone should be gone as well)

[binary.zip](https://github.com/microsoft/pxt-adafruit/files/3214473/binary.zip)

Jaqster, 2019-05-24:
No, I don't hear any improvement.  These are v1.22 firmware boards I'm testing on.  Also the Simulator doesn't sound great on the computer either...

abchatra, 2019-05-28:
Did this bug got fixed?

mmoskal, 2019-05-28:
I don't hear the buzzy sound...

Jaqster, 2019-05-28:
I think the Play Tone blocks now sound less buzzy than before.  But the weird sound between PlaySoundUntil Done blocks is still there, and I'm not sure the Volume is correct - 255 still sounds quiet to me.

abchatra, 2019-06-11:
@Jaqster playTone volume is still low for default and doesn't sound good.

abchatra, 2019-08-15:
We have new issues, but closing this old one.

closed 2019-08-15 by abchatra