microsoft/pxt-adafruit #1044: Redundant Text from Minerva in SIREN tutorial

kimprice, 2019-05-23:
**Describe the bug**
Minerva speech bubble has redundant text to the instructions. The speech bubble opens right when the tutorial starts.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://makecode.adafruit.com/beta
2. Click on Siren tutorial.

**Expected behavior**
I expected Minerva to have an intro speech bubble like "Hi, I'm Minerva. Click me if you need help." I don't think Minerva's speech should be the same as the instructions.

**Screenshots**
[image]

kimprice, 2019-05-23:
This is happening for every step in this tutorial. When clicking the next step, the speech bubble automatically appears and has the exact same text as the instructions.

abchatra, 2019-08-15:
Fixed

closed 2019-08-15 by abchatra