microsoft/pxt-adafruit #1061: analog write errors on Circuit Playground Express A0 (true analogue)

kevinjwalters, 2019-05-28:
**Describe the bug**
Very simple programs using `analog write` block to set an analogue value on a Circuit Playground Express board's **A0** pin/pad (true analogue, not PWM) error immediately with a `909: PANIC_INVALID_ARGUMENT): the argument value is out of range or the type or format is invalid` as indicated by colourful red/blue NeoPixels. D13 flashes too.

**To Reproduce**
Steps to reproduce the behavior:
1. Use analog write to pin A0

**Expected behavior**
A constant voltage appear on the `A0` pad without error.

**Screenshots**
See code on https://makecode.com/_RDm03xPfVdz1

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Firefox and Chrome
 - Version: n/a (?)

**Additional context**
I see there's a discussion about `analog write` on https://github.com/microsoft/pxt-adafruit/issues/604 and in particular a mention of `A0` and "Michal's fixes" in https://github.com/microsoft/pxt-adafruit/issues/604#issuecomment-381021875 from Apr 2018.

kevinjwalters, 2019-05-28:
I tried [MakeCode beta](https://makecode.adafruit.com/beta#editor) on 28-May-2019 and same issue there.

```
Adafruit Circuit Playground Express version:  1.4.28
Microsoft MakeCode version:  5.13.6
codal-circuit-playground runtime version:  v1.5.4-i2c_and_dac.0
```

abchatra, 2019-05-29:
@pelikhan @mmoskal

kevinjwalters, 2019-08-28:
Is there any ETA on this?

kevinjwalters, 2020-06-22:
I saw someone mention a "strange result" from analogue writes to **A0** in [AnalogWrite in MS MakeCode for pins A3, A6, A7](https://forums.adafruit.com/viewtopic.php?f=58&t=166491), I'll ask what this was...

smagoun, 2021-01-07:
I can reproduce the analogWrite problem with analog writes on A0 using the program in the original report. It's reproducible in both MakeCode + MakeCode beta. The error code on mine is b0001011010 / 90 (PANIC_CODAL_HARDWARE_CONFIGURATION_ERROR) rather than 909.

MakeCode:
Adafruit Circuit Playground Express version:  1.4.43
Microsoft MakeCode version:  5.20.6
codal-circuit-playground runtime version:  v2.0.1

Makecode Beta:
Adafruit Circuit Playground Express version:  1.6.4
Microsoft MakeCode version:  6.1.10
codal-circuit-playground runtime version:  v2.0.4

smagoun, 2021-01-09:
I did some more sleuthing. The panic is a regression between 1.2.13 and 1.3.1.

https://makecode.adafruit.com/v1.2.13#editor - code runs:
Adafruit Circuit Playground Express version:  1.2.13
Microsoft MakeCode version:  5.1.5
codal-circuit-playground runtime version:  v1.5.6

https://makecode.adafruit.com/v1.3.1#editor - same code causes panic:
Adafruit Circuit Playground Express version:  1.3.1
Microsoft MakeCode version:  5.5.25
codal-circuit-playground runtime version:  v1.5.4-samd-peripherals.10