microsoft/pxt-adafruit #1081: Some pictures in the tutorial "make guitar" can't be loaded

Helen09, 2019-08-19:
**Describe the bug**
Some pictures in the tutorial "make guitar" can't be loaded

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/projects/guitar/make
2. Scroll down to bottom of tutorial

**Expected behavior**
Picture in the tutorial should can be loaded

**Actual behavior**
Some pictures in the tutorial "make guitar" can't be loaded
[image]
[image]
[image]

**Additional info:**
1. OS: Windows (rs6)
2. Adafruit Circuit Playground Express version: 1.4.31
3. Microsoft MakeCode version: 5.20.3
4. codal-circuit-playground runtime version: v2.0.0

ganicke, 2019-08-19:
I'm seeing the same problem, this works on localserve though. Hmm, images are all here:

https://github.com/microsoft/pxt-adafruit/tree/master/docs/static/cp/projects/guitar

ganicke, 2020-04-13:
Images are serving, closing this.

closed 2020-04-13 by ganicke