microsoft/pxt-adafruit #1087: Tempo Doesn't Appear

dharinirajah, 2019-08-19:
**Describe the bug**
The tempo isn't appearing in the melody editor but shows up in the exterior block.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://makecode.adafruit.com/beta#editor
2. Drag out a melody editor block
3. See error

**Expected behavior**
You should be able to see the tempo next to the play button

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Firefox

dharinirajah, 2019-08-19:
Text appears to be white as when highlighted the number appears.
[image]

jwunderl, 2019-08-19:
will be fixed by https://github.com/microsoft/pxt/pull/5826

abchatra, 2019-08-29:
fixed

closed 2019-08-29 by abchatra