microsoft/pxt-adafruit #1095: Download and serial stream doesn't work on window10 app

abchatra, 2019-08-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install MakeCode for adafruit windows 10 app
2. Create a project
3. Click on download
4. See error

**Expected behavior**
Drag and drop not required to download in windows 10 app. Also console.log data doesn' seem to flow back to the app

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 
 - MakeCode for Adafruit app.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

darzu, 2019-08-21:
might be the same issue as https://github.com/microsoft/pxt-adafruit/issues/991

abchatra, 2019-08-29:
Fixed

closed 2019-08-29 by abchatra