microsoft/pxt-adafruit #1111: The first digit of tempo can't be deleted in block "play melody"

Helen09, 2019-08-30:
**Describe the bug**
The first digit of tempo can't be deleted in block "play melody" 

**Steps to reproduce the behavior**
1. Navigate to https://makecode.adafruit.com/beta#editor
2. Click “MUSIC” -> drag out block "play melody" and put it under "on start"
3. Try to change the tempo number to delete it original number

**Expected behavior**
The first digit of tempo should can be deleted in block "play melody" 

**Actual behavior**
The first digit of tempo can't be deleted in block "play melody" 
[image]

**Additional info:**
1. OS: Windows (rs6)
2. Adafruit Circuit Playground Express version: 1.4.34
3. Microsoft MakeCode version: 5.20.4
4. codal-circuit-playground runtime version: v2.0.0

abchatra, 2019-08-30:
By design, can't have lower than 1 bpm

Helen09, 2019-09-02:
But when user want to input other numbers for example "80", "95" the "1" can't be deleted

abchatra, 2019-09-02:
Valid issue