microsoft/pxt-adafruit #1167: Servo Extension blocks broken

Jaqster, 2020-04-30:
Hey Peli,
I think your Servo motor blocks are broken.
This doesn't work.  But the Servo Pin blocks do.
[image]

pelikhan, 2020-07-14:
I think You need to remove the stop servo blocks

Jaqster, 2020-07-15:
you're right.  When I remove the Stop Servo blocks, it works.  
This was the hardware error I was getting with the Stop Servo blocks (I like these though, since they stop the buzzing/jittering)
[image]

pelikhan, 2020-07-15:
Add a pause 1s before stop

[Servo error]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F22108954%2F87489741-9f190600-c5f8-11ea-86ef-b07ebbb30b42.jpg&data=02%7C01%7Cjhalleux%40microsoft.com%7C6c29a6e0d3ad4682012108d828573da0%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637303702446158699&sdata=smyFaJqcInYQJe3ckRQncgSI6NQ4slFJxAaBjQljsEQ%3D&reserved=0>